sleemanj/DS3231_Simple #2: Not erasing logs after reading them

Cyberteen, 2017-02-01:
Hi,
Thanks for this great library.

Is it possible to possible to preserve the logs after reading them using readLog function?
Maybe disable the makeEEPROMSpace function in the readLog() to do that.

Thanks

sleemanj, 2017-02-01:
Yes you could as you say remove the call to `makeEEPROMSpace` this will prevent the just-read log entry from being killed.  

To read that log entry again you would then have to read all the other log entries one by one to "walk along the log" as it were until you get back to that one (or implement some sort of rewind/seek function).

You would want your log data to include some sort of identifier so that you knew, when reading, that you had got to the start again, otherwise you could be reading logs in a loop forever :-)

Note also that the nature of a circular buffer is that old log entries will be over-written by new log entries when the buffer is full.

closed 2017-02-01 by sleemanj

geologic, 2017-07-17:
I also need to preserve logs after readLog(). 
I commented makeEEPROMSpace line but everytime i do a readLog(), i only see the logs between readLog() calls. What else do i need to modify to everytime i call readLog() i get all logs from the beginning?