sleemanj/DS3231_Simple #3: Wake Arduino from Sleep

zstergios, 2017-04-13:
I have wrote a test sketch https://ufile.io/989881

It supposed that the DS3221 it will powered by the coin battery and not permanlty from the arduino power.

So VCC connected to PIN 13
` pinMode(13, OUTPUT);
  digitalWrite(13, HIGH);
  Clock.begin();
`

When the PIN-13 goes LOW, DS3221 does not send any signal (interrupt) to arduino.

I have connected to UNO:
VCC to PIN 13
GND to GND
SDA to A4
SCL to A5 

Seems like not keeps in memory the alarms?
Any help?

sleemanj, 2017-04-13:
If you set Pin 13 to Low, you are probably short-circuiting the battery on the clock module.

Send Pin 13 to high impedance state (INPUT), not low.

closed 2017-04-13 by sleemanj

zstergios, 2017-04-15:
I don't think it causes shot-circuit. Module is compatible with v3/v5 logic.
Library works great without sleep.

But when the arduino sleeps, SQW (connected to PIN2) works only with permanent power supply of arduino.

I'm reading over the Internet, and says to set to 1HZ the SQW, I don't know how to do that with the library.

What I want:
1)setAlarm
2) Sleep, then the DS3231 will be powered by coin battery
3) DS3231 wakes arduino, and is powered by arduino VCC (I think it's the only way to send commands to RTC)

geologic, 2017-07-17:
You need to do some dessoldering if you want to use alarms with coin cell power:
https://edwardmallon.wordpress.com/2014/05/21/using-a-cheap-3-ds3231-rtc-at24c32-eeprom-from-ebay

You just need to remove the resistor array, but it is goog idea to get rid of the led and charging circuit

zstergios, 2017-07-31:
Thank's mate!

zstergios, 2017-11-20:
The solution should be added to librart @sleemanj 
http://prntscr.com/hcst87
Comment from https://edwardmallon.wordpress.com/2014/05/21/using-a-cheap-3-ds3231-rtc-at24c32-eeprom-from-ebay/comment-page-1/#comment-2431

zstergios, 2018-02-19:
Can you please add function to squareWave enable/disable?

There is already implemented to [other library](https://github.com/JChristensen/DS3232RTC), but I would like to work with yours