sleemanj/DS3231_Simple #9: Daylight saving Time

bobybobybob, 2018-01-22:
Hello

very great Programm, i like it very much. I just miss one thing... that annoying summer- and wintertime. 

what do you recommend for this problem? Can you make a sample program?

sleemanj, 2018-01-22:
DST is timezone dependant.  you need to adjust the time manually to suit your specific place in the world.

I recommend to always deal only in UTC (GMT, Time Zulu) then you don't need to worry about such things.

See Tom Scott's video for why it's not a simple problem
https://www.youtube.com/watch?v=-5wpm-gesOY

closed 2018-01-22 by sleemanj

bobybobybob, 2018-01-22:
Hallo and thank you very much for fast reaction and answer. 

UTC time is not used in my living environment. It also confuses having two different time formats. (Local and UTC). 

Christiansen has a library called Timezone. It is also possible to give the UTC time that libary and it then says if now summer time or not. It work, but is in your RTC library the possible to changing the current time by +/- X seconds during operation?  RTC have UTC Time but output is Local Time with custom +/- Seconds or set the RTC with +/- X Seconds. 

That would be easy to make a Word Clock too.

bobybobybob, 2018-01-26:
may God bless you on all your ways. Haleluja i solve the problem

zstergios, 2018-02-19:
how you solved the time changes ?

bobybobybob, 2018-02-19:
RTC in UTC Time and local Time for use with Timezone Lib. Set the Timezonelib with the UTC Time. For Alarm i have to use UTC Time so i use a if for alarm in wintertime and one für summertime. 

Another way is use Timezonelib and RTC in local time. The RTC Lib + Flag sets the RTC Time new. 

Both are ugly and not nice, but work.  I prever the first one