sleemanj/DS3231_Simple #11: Wire.requestFrom() should not be followed by Wire.endTransmission().

Koepel, 2018-03-17:
In the file "DS3231_Simple.cpp", in the function "readEEPROMByte", there is Wire.endTransmission() after the Wire.requestFrom(). That Wire.endTransmission() is currently in the line 125 and can be removed.

The Wire.endTransmission() should only be used when writing data. It does not just send a stop, it does not just finish something, it can not correct something that went wrong. It should only be used when writing data and that is all it is supposed to do.

sleemanj, 2018-03-17:
On this occasion you are not correct.

To quote from the datasheet ( http://www.switchdoc.com/wp-content/uploads/2015/01/doc0336.pdf )

> STOP CONDITION: A low-to-high transition of SDA with SCL high is a stop condition.
**After a read sequence, the stop command will place the EEPROM in a standby power
mode (refer to Start and Stop Definition timing diagram)**

> RANDOM READ: A random read requires a “dummy” byte write sequence to load in the data
word address. **Once the device address word and data word address are clocked in and
acknowledged by the EEPROM, the microcontroller must generate another start condition**

And further figure 5 of the datasheet which illustrates the above.

You can see in the code that on line [115](https://github.com/sleemanj/DS3231_Simple/blob/master/DS3231_Simple.cpp#L115) endTransmission is used to send the repeated start after the dummy write, by setting the `false` parameter it causes Wire library to NOT send the STOP, **in accordance with the datasheet**.

On line 125 the endTransmission is called (with default parameter `true`) to now send the STOP, again in accordance with the datasheet.

closed 2018-03-17 by sleemanj

Koepel, 2018-03-17:
This is about the way the Wire library is implemented. The Wire.endTransmission() can not be used to stop or not stop something. That is not what it does.

I say it again: "_The Wire.endTransmission() should only be used when writing data. It does not just send a stop, it does not just finish something, it can not correct something that went wrong. It should only be used when writing data and that is all it is supposed to do._"

Koepel, 2018-03-26:
Hi, are you willing to look into it and adjust your code to make it according to the datasheet ?
I have made a wiki to explain the usage of the Wire library: [Common-mistakes#2](https://github.com/Koepel/How-to-use-the-Arduino-Wire-library/wiki/Common-mistakes#2)

Koepel, 2019-05-06:
The wrong `Wire.endTransmission()` is still there. At the moment at line 124. You may remove that line.