sleemanj/DS3231_Simple #13: Set alarm every X minutes

fedcas, 2018-07-11:
Hi,
thanks for your library, it's the one that I'm using ;) there's just a couple of things I feel the need for.

I understand I can I can set these alarms:

ALARM_EVERY_MINUTE [fires every minute]
ALARM_MATCH_SECOND [fires every minute at a specified second]
ALARM_HOURLY [fires every minute]
ALARM_MATCH_MINUTE [fires every minute at a specified minute]

But how can I, for example, set an alarm every 5 minutes, 10 minutes, 15 minutes or 30 minutes? I'm using your library for a datalogger, and I would like to increase the logging interval. I'm using a counter for the moment, but hopefully there's a better way.

Thanks

geologic, 2018-07-12:
The only way is using a counter.
For 15 minutes, you set an ALARM_EVERY_MINUTE and increment a counter. When counter=15, thats your 15 minute interval.

Experiment-6-2-6, 2018-07-13:
You can use ALARM_MATCH_MINUTE_HOUR.
First you have to check and save the time then the alarm starts.
Next you adding your time (for example 5 or 30 minutes but mind you sometime have to can change the hour)
Next you set second alarm clock to match that time.

fedcas, 2018-07-17:
thanks ;) I think I like your approach better than what I was doing with the counter. The only thing is, isn't it more simple to use just the minutes?

here is the code that I'm trying at the moment, it seems to work:

```c
[...]
    int interval = 2; //minutes
    DateTime time;
    int mins;
[...]

void setup() {
[...]
  time = Clock.read();
  mins = time.Minute;
  
  mins = mins + interval;
  if (mins >= 60) mins = mins - 60;
  time.Minute = mins;
  Clock.setAlarm(time, DS3231_Simple::ALARM_MATCH_MINUTE);
[...]
}

void loop() 
{ 
  if(Clock.checkAlarms())
  {
[...]
    mins = mins + interval;
    if (mins >= 60) mins = mins - 60;
    time.Minute = mins;
    Clock.setAlarm(time, DS3231_Simple::ALARM_MATCH_MINUTE);
[...]
  }
}
```

Experiment-6-2-6, 2018-07-17:
Well, that is better than my idea (of course if the interval is less than 59 minutes)
By the way, I found this article very instructive about limitation of DS3231 alarms.
https://gist.github.com/JChristensen/0359516e3780a819cbffef0db5419213

fedcas, 2018-07-18:
two heads are better than one ;) I'll read the article too, thanks ;)

CaptChilko, 2019-05-05:
Thanks, this is exactly what I'm looking for!

Just a note for future visitors/users, `if (mins >= 60) mins = mins - 60;` can be replaced with `mins = mins % 60;`