sleemanj/DS3231_Simple #20: PauseClock option in checkAlarms uses wrong register

oschonrock, 2020-07-07:
```C++
if(PauseClock)                                                                                                                                                                      
  {                                                                                                                                                                                   
    if(rtc_i2c_read_byte(0xE,StatusByte))
```

But it should be 0x0F:
https://datasheets.maximintegrated.com/en/ds/DS3231.pdf
```
Status Register (0Fh)
Bit 7: Oscillator Stop Flag (OSF).
....
```

sleemanj, 2020-07-07:
Been too many years since I wrote this to remember, but 

There are 2 alarms, eahch has it's own bit in the status register (which 0x3 covers both for the bitwise comparison) so the race condition possible is...

   1. Alarm 1 Fires and bit 0 is set in the status register
   2. Read status variable and see that one of bit 0 or bit 1 was set so alarm was fired
   3. Alarm 2 Fires and bit 1 is set in the status register
   4. Clear Status, both bit 0 and 1 are cleared
   5. Return the status we read, which was Alarm 1 fired, bit 0 set, but not bit 1, Alarm 2 is lost.

A quick skim (again long long time since I wrote this) says that 0X0E (page 13) is the control register bit 7 is what we WANT the clock to be enabled or disabled, while 0X0F (page 14)  is the status register which tells us if it is enabled/disabled (which it may be disabled even if we wanted it enabled, for various reasons).

I think that the code is correct to write to the control register (E) bit 7, although, does rely on the clock running exclusively from battery (page 13, bit 7 details). It's not clear to me from the datasheet if it's ok to write 1 to the status register (F).

oschonrock, 2020-07-07:
Re 2 alarms. I think, I updated my above report concurrently with your answer. But again briefly here:

I do **not** believe a race condition is possible even with 2 alarms, because the alarms bits inside the DS3231 are triggered by matching the current time (minutes, seconds etc) against the alarm time via a mask . This is not spelled out in the Datasheet, but as a reasonable "assumption" (dangerous word, I know) the DS3231 will change both alarm bits at the same time, ie when the seconds match the alarm time - on that very internal clock pulse the 2 flip-flips on the silicon will change. The A1F and A2F flasg can only change once per second, by definition. If the Alarms are both set to same time, then will either both or neither become set on the same internal DS3231 clock. 

So, no, I do not believe that it is possible for you to get "in between 2 internal DS3231 clock pulses" with your I2C request. 

Regarding the Oscillator enable:
You're right regarding Register 0x0F. I misread that on skimming. It's an indicator, not a control bit.

However, as you said 0x0E is for VBAT only:
```
Control Register (0Eh)
Bit 7: Enable Oscillator (EOSC). When set to logic 0,
the oscillator is started. When set to logic 1, the oscillator
is stopped when the DS3231 switches to VBAT. This bit
is clear (logic 0) when power is first applied. When the
DS3231 is powered by VCC, the oscillator is always on
regardless of the status of the EOSC bit. When EOSC is
disabled, all register data is static.

```
However, checkAlarms() is for **polling**, ie we have chosen that we do not want to use a hardware interrupt. Under what circumstances might we be polling the DS3231 when it is on VBAT? What are we polling it with? An MCU which is also on battery? And the DS3231 consumes so much power that we shut it down, but not the arduino MCU? 

So I think my original conclusion is probably still correct:

1. There is no race condition - (Also, No other DS3231 lib that I am aware of attempts to deal with it...)

2. Setting 0x0E.7 to 1 will not stop the clock, because the DS3231 will not be on VBAT, because we are talking to it from an ardudino which clearly has Vcc and so will the clock - It would be a very very strange application, where that is not true. 

So - symptom doesn't exist, and the treatment is ineffective.

sleemanj, 2020-07-07:
1. Alarm 1 fires and bit set at Second N
  2. At Second N+(999999 microseconds) we read status
  3. At Second N+1 Alarm 2 fires and bit set
  4. At Second N+1+(1 microsecond) we clear status

You are free to fork as you disagree, I however and closing wontfix.

closed 2020-07-07 by sleemanj

sleemanj, 2020-07-07:
> if alarm1 = HH:MM:59 and alarm2=HH:MM+1:00

The race condition will occur when the second alarm (be it alarm 1 or alarm 2) fires after the poll but before the state is updated.    The first alarm to fire could be ages before you poll. The alarms do not need to fire 1 second apart to cause the race.

Yes it is a very small window of time, microseconds usually but because checkAlarms does not disable interrupts, potentially variable and not insignificant.

Conversation ends.