sleemanj/DS3231_Simple #23: Error Compiling

cmjones1234, 2022-03-09:
I am newer to programming and have successfully uploaded the code to a Nano for an LED Display clock.  I am now trying to accomplish the same thing via an ESP8266 ESP-12F (WeMos D1 Mini) and am having issues with Verifying/Uploading.  I keep getting the error message below and am wondering if I need to add a different code or a library, as I would like to keep the same project coding process, just with the ability to adjust the color scheme/set alarms through WiFi.  any help would be appreciated.

Setting:
Arduino: 1.8.16 (Windows 10), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Disabled (new aborts on oom), Disabled, All SSL ciphers (most compatible), 32KB cache + 32KB IRAM (balanced), Use pgm_read macros for IRAM/PROGMEM, 4MB (FS:2MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Messages:
C:\Users\Chadwick\Documents\Arduino\libraries\DS3231_Simple\DS3231_Simple.cpp: In member function 'DS3231_Simple::DateTime DS3231_Simple::read()':
C:\Users\Chadwick\Documents\Arduino\libraries\DS3231_Simple\DS3231_Simple.cpp:414:12: error: '_BV' was not declared in this scope
  414 |     if(x & _BV(6))
      |            ^~~
C:\Users\Chadwick\Documents\Arduino\libraries\DS3231_Simple\DS3231_Simple.cpp:428:10: error: '_BV' was not declared in this scope
  428 |     if(x&_BV(7))
      |          ^~~
C:\Users\Chadwick\Documents\Arduino\libraries\DS3231_Simple\DS3231_Simple.cpp: In member function 'uint8_t DS3231_Simple::setAlarm(const DS3231_Simple::DateTime&, uint8_t)':
C:\Users\Chadwick\Documents\Arduino\libraries\DS3231_Simple\DS3231_Simple.cpp:491:33: error: '_BV' was not declared in this scope
  491 |     controlByte = controlByte | _BV(0) | _BV(2); // Enable Alarm 1, set interrupt output on alarm.
      |                                 ^~~
C:\Users\Chadwick\Documents\Arduino\libraries\DS3231_Simple\DS3231_Simple.cpp:496:33: error: '_BV' was not declared in this scope
  496 |     controlByte = controlByte | _BV(1) | _BV(2); // Enable Alarm 2, set interrupt output on alarm.
      |                                 ^~~
C:\Users\Chadwick\Documents\Arduino\libraries\DS3231_Simple\DS3231_Simple.cpp:508:70: error: '_BV' was not declared in this scope
  508 |     Wire.write(bin2bcd(AlarmDate.Dow)   | (AlarmMode & 0B10000000) | _BV(6));
      |                                                                      ^~~
C:\Users\Chadwick\Documents\Arduino\libraries\DS3231_Simple\DS3231_Simple.cpp: In member function 'uint8_t DS3231_Simple::checkAlarms(uint8_t, uint8_t)':
C:\Users\Chadwick\Documents\Arduino\libraries\DS3231_Simple\DS3231_Simple.cpp:537:44: error: '_BV' was not declared in this scope
  537 |       rtc_i2c_write_byte(0xE, StatusByte | _BV(7));
      |                                            ^~~
C:\Users\Chadwick\Documents\Arduino\libraries\DS3231_Simple\DS3231_Simple.cpp:556:46: error: '_BV' was not declared in this scope
  556 |       rtc_i2c_write_byte(0xE, PauseClock & ~(_BV(7)));
      |                                              ^~~
exit status 1
Error compiling for board LOLIN(WEMOS) D1 R2 & mini.

sleemanj, 2022-03-09:
Your board's core does not have the _BV macro it seems.  You could add it to the DS3231_Simple.h header fileHere it is in the esp8266 core...https://github.com/esp8266/Arduino/commit/22f063b913884f0ee96ef3cd2495e4979f7f6d92 ---- On Thu, 10 Mar 2022 08:55:35 +1300  ***@***.***  wrote ----
I am newer to programming and have successfully uploaded the code to a Nano for an LED Display clock.  I am now trying to accomplish the same thing via an ESP8266 ESP-12F (WeMos D1 Mini) and am having issues with Verifying/Uploading.  I keep getting the error message below and am wondering if I need to add a different code or a library, as I would like to keep the same project coding process, just with the ability to adjust the color scheme/set alarms through WiFi.  any help would be appreciated.
Setting:
Arduino: 1.8.16 (Windows 10), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Disabled (new aborts on oom), Disabled, All SSL ciphers (most compatible), 32KB cache + 32KB IRAM (balanced), Use pgm_read macros for IRAM/PROGMEM, 4MB (FS:2MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 115200"
Messages:
C:\Users\Chadwick\Documents\Arduino\libraries\DS3231_Simple\DS3231_Simple.cpp: In member function 'DS3231_Simple::DateTime DS3231_Simple::read()':
C:\Users\Chadwick\Documents\Arduino\libraries\DS3231_Simple\DS3231_Simple.cpp:414:12: error: '_BV' was not declared in this scope
414 |     if(x & _BV(6))
|            ^~~
C:\Users\Chadwick\Documents\Arduino\libraries\DS3231_Simple\DS3231_Simple.cpp:428:10: error: '_BV' was not declared in this scope
428 |     if(x&_BV(7))
|          ^~~
C:\Users\Chadwick\Documents\Arduino\libraries\DS3231_Simple\DS3231_Simple.cpp: In member function 'uint8_t DS3231_Simple::setAlarm(const DS3231_Simple::DateTime&, uint8_t)':
C:\Users\Chadwick\Documents\Arduino\libraries\DS3231_Simple\DS3231_Simple.cpp:491:33: error: '_BV' was not declared in this scope
491 |     controlByte = controlByte | _BV(0) | _BV(2); // Enable Alarm 1, set interrupt output on alarm.
|                                 ^~~
C:\Users\Chadwick\Documents\Arduino\libraries\DS3231_Simple\DS3231_Simple.cpp:496:33: error: '_BV' was not declared in this scope
496 |     controlByte = controlByte | _BV(1) | _BV(2); // Enable Alarm 2, set interrupt output on alarm.
|                                 ^~~
C:\Users\Chadwick\Documents\Arduino\libraries\DS3231_Simple\DS3231_Simple.cpp:508:70: error: '_BV' was not declared in this scope
508 |     Wire.write(bin2bcd(AlarmDate.Dow)   | (AlarmMode & 0B10000000) | _BV(6));
|                                                                      ^~~
C:\Users\Chadwick\Documents\Arduino\libraries\DS3231_Simple\DS3231_Simple.cpp: In member function 'uint8_t DS3231_Simple::checkAlarms(uint8_t, uint8_t)':
C:\Users\Chadwick\Documents\Arduino\libraries\DS3231_Simple\DS3231_Simple.cpp:537:44: error: '_BV' was not declared in this scope
537 |       rtc_i2c_write_byte(0xE, StatusByte | _BV(7));
|                                            ^~~
C:\Users\Chadwick\Documents\Arduino\libraries\DS3231_Simple\DS3231_Simple.cpp:556:46: error: '_BV' was not declared in this scope
556 |       rtc_i2c_write_byte(0xE, PauseClock & ~(_BV(7)));
|                                              ^~~
exit status 1
Error compiling for board LOLIN(WEMOS) D1 R2 & mini.

sleemanj, 2022-03-09:
https://github.com/sleemanj/DS3231_Simple/commit/ab8e17fd2ee30fd492eec4d1006606e9e9b652cb

Closes

closed 2022-03-09 by sleemanj

cmjones1234, 2022-03-10:
Thank you for the work put in. Being newer to programming, I was still unable to get it to load up using the last posting you listed. I will continue to try adding it to the original DS3231_Simple or removing all instances of it and reloading the .zip of new version you posted. Sorry for not being as familiar as I should with IDEs but I am learning a lot. If you have any suggestions for the "layman" I would be receptive. My goal is to attempt this clock I just made (props to Lewis for https://youtu.be/8E0SeycTzHw) as a wifi accessible clock now!Sent via the Samsung Galaxy A11, an AT&T 4G LTE smartphone

Closed #23.

sleemanj, 2022-03-10:
Locate the file on your system "DS3231_Simple.h".

Open the file in text editor of your choice

Add at top of file

    #ifndef _BV
    #define _BV(b) (1UL << (b))
    #endif

cmjones1234, 2022-10-11:
Thank you for the work put in. Being newer to programming, I was still unable to get it to load up using the last posting you listed. I will continue to try adding it to the original DS3231_Simple or removing all instances of it and reloading the .zip of new version you posted. Sorry for not being as familiar as I should with IDEs but I am learning a lot. If you have any suggestions for the "layman" I would be receptive. My goal is to attempt this clock I just made (props to Lewis for https://youtu.be/8E0SeycTzHw) as a wifi accessible clock now!Sent via the Samsung Galaxy A11, an AT&T 4G LTE smartphone