sleemanj/DS3231_Simple #25: No longer working - seemingly due to an Arduino IDE update sometime in the last 12 months.

fkkjohn, 2023-04-10:
Almost exactly one year ago I used this library to make a 30 minute timer to automatically turn off a relay when the alarm triggers. It worked perfectly, I saved the Arduino sketch. Now I am attempting to build another such timer, and I have run into trouble. Here's what happened. I opened my saved sketch and attempted to build and upload to the exact same type of microcontroller (ATTiny85). It failed to compile. The errors were mostly related to "Stream" not being declared in ds3231_simple.cpp. Line 19, "void DS3231_Simple::print_zero_padded(Stream &Printer, uint8_t x)", seemed to cause the most trouble. There were a few other smaller errors, but they all went away at the same time. 

I "_fixed_" the problem after deleting my Arduino15 folder under /AppData/Local. It was a last ditch attempt since nothing else was getting rid of the errors. Once deleted, I started Arduino IDE again. The IDE rebuilt everything that it needed that was missing from that deleted directory, and when I tried to upload the sketch, it compiled just fine. It even behaves correctly in everyway except things related to the DS3231. The power light on the RTC module is on, and it's wired exactly the same way as the working timer, but it never triggers the alarm. I want to debug this futher, but I'm not currently set up for it, and I could probably switch libraries faster. I verified that it IS the library that's broken by uploading the successfully compiled sketch to my older and working timer. Sure enough, same behavior. Everything is fine except the RTC. Same microcontroller, same RTC module, the only difference is the Arduino IDE, which apparently went through some "update" that broke everything. Now it even seems like it isn't broken, but it definitely still is.

I'd love to use this library again, because it's dead simple to get to work, but I'm now forced to move on.

sleemanj, 2023-04-10:
Ensure you are using an appropriate core. 

[ATTinyCore 1.5.x, not 2.x which is still under development](https://github.com/SpenceKonde/ATTinyCore)

[My fork of ATTinyCore][https://github.com/sleemanj/optiboot/blob/master/dists/README.md)

 The standard Arduino core would not be appropriate for a Tiny85.  Use ATTinyCore or my own fork of it (

fkkjohn, 2023-04-11:
Thank you very much for your reply. I'm currently removing all the updated Arduino IDE software and reverting to my older setup. I'll definitely try both ATTinyCores. I rewrote my program using the DS3231 (no affixes) library and the TinyWireM library, and it's much worse as it seems to be overwriting the bootloader as well. I'll let you know how things go so maybe others who run into the same problem might find some sliver of pertinent information on their Google searches, unlike my searches yesterday.

fkkjohn, 2023-04-11:
I'm still having trouble, and it varies per version of the Arduino IDE, and which programmer/board I use.

Here's what I was getting at first when trying to recompile this sketch that worked 1 year prior. I just got them all again using your fork of the ATTinyCore, your DS3231 Simple (same zip i got from github 1 year ago), with Arduino 1.8.18 (newer version as you recommend... I'm not saying this is your fault, I'm convinced this is due to the IDE, but by using all your tools, maybe you can rule your stuff out or tell me what I'm missing). I also tried it with IDE 1.6.4 and had nonstop trouble with Wire.h not found:

Arduino: 1.8.18 (Windows 10), Board: "ATtiny85, Yes (Normal Arduino Serial/USB Upload), 8MHz Internal Oscillator, Millis and Tone Available, Bin, Hex, Dec Supported, Better Or Equal 1.666% Error (Default), LTO Enabled, 2.7v, Default, Default, Default"

```
In file included from C:\Users\owner\Documents\Arduino\libraries\DS3231_Simple-master\DS3231_Simple.cpp:1:0:

C:\Users\owner\Documents\Arduino\libraries\DS3231_Simple-master/DS3231_Simple.h:88:38: error: 'Stream' has not been declared

     static void    print_zero_padded(Stream &Printer, uint8_t x);

                                      ^~~~~~

C:\Users\owner\Documents\Arduino\libraries\DS3231_Simple-master/DS3231_Simple.h:148:35: error: 'Stream' has not been declared

     void     promptForTimeAndDate(Stream &Serial);

                                   ^~~~~~

C:\Users\owner\Documents\Arduino\libraries\DS3231_Simple-master/DS3231_Simple.h:252:22: error: 'Stream' has not been declared

     void     printTo(Stream &Printer);

                      ^~~~~~

C:\Users\owner\Documents\Arduino\libraries\DS3231_Simple-master/DS3231_Simple.h:263:22: error: 'Stream' has not been declared

     void     printTo(Stream &Printer, const DateTime &Timestamp);

                      ^~~~~~

C:\Users\owner\Documents\Arduino\libraries\DS3231_Simple-master/DS3231_Simple.h:268:30: error: 'Stream' has not been declared

     void     printDateTo_DMY(Stream &Printer, const DateTime &Timestamp, const char separator = '/');

                              ^~~~~~

C:\Users\owner\Documents\Arduino\libraries\DS3231_Simple-master/DS3231_Simple.h:273:30: error: 'Stream' has not been declared

     void     printDateTo_DMY(Stream &Printer) { printDateTo_DMY(Printer, read()); }

                              ^~~~~~

C:\Users\owner\Documents\Arduino\libraries\DS3231_Simple-master/DS3231_Simple.h:278:30: error: 'Stream' has not been declared

     void     printDateTo_MDY(Stream &Printer, const DateTime &Timestamp, const char separator = '/');

                              ^~~~~~

C:\Users\owner\Documents\Arduino\libraries\DS3231_Simple-master/DS3231_Simple.h:283:30: error: 'Stream' has not been declared

     void     printDateTo_MDY(Stream &Printer) { printDateTo_MDY(Printer, read()); }

                              ^~~~~~

C:\Users\owner\Documents\Arduino\libraries\DS3231_Simple-master/DS3231_Simple.h:288:30: error: 'Stream' has not been declared

     void     printDateTo_YMD(Stream &Printer, const DateTime &Timestamp, const char separator = '-');

                              ^~~~~~

C:\Users\owner\Documents\Arduino\libraries\DS3231_Simple-master/DS3231_Simple.h:293:30: error: 'Stream' has not been declared

     void     printDateTo_YMD(Stream &Printer) { printDateTo_YMD(Printer, read()); }

                              ^~~~~~

C:\Users\owner\Documents\Arduino\libraries\DS3231_Simple-master/DS3231_Simple.h:298:30: error: 'Stream' has not been declared
[... 462 lines cut ...]

C:\Users\owner\Documents\Arduino\libraries\DS3231_Simple-master\DS3231_Simple.cpp:743:42: note: suggested alternative: 'fread'

 void DS3231_Simple::promptForTimeAndDate(Stream &Serial)

                                          ^~~~~~

                                          fread

C:\Users\owner\Documents\Arduino\libraries\DS3231_Simple-master\DS3231_Simple.cpp:743:50: error: 'Serial' was not declared in this scope

 void DS3231_Simple::promptForTimeAndDate(Stream &Serial)

                                                  ^~~~~~

exit status 1

Error compiling for board ATtiny85.
This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

I'm going to keep at it till I get it working again, but this was exactly my first set of errors. I have removed all previous versions of the IDE and the libraries, etc, and installed it all from scratch using what I think I had available to me at the time. I'll include my sketch in a second comment so you can look at that. It hasn't been changed since it worked.

fkkjohn, 2023-04-11:
I have a digispark (branded as one anyway) rev3 connected to a MW-084 labeled DS3231 RTC, an active buzzer (5v), and 4v input SSD relay. The program turns on the relay on boot, starts a (for testing purposes) 1 min timer, after the alarm goes off, it turns off the relay after flashing it a few times and sounding the buzzer. When you press the attached momentary switch, it resets the timer (this time to the desired 30 min duration) and it all starts again. I rewrite the time to the RTC to give me a simple reset point so I don't have to deal with millis or writing a bunch of date/time settings to variables. It's about as simple and as basic as they come, so I was really shocked when it suddenly wouldn't compile anymore. I wish the Arduino IDE had a way to save the binary files and archive them for future uploading instead of having to rebuild something with constantly evolving libraries and such. What a PITA.

```
#include <DS3231_Simple.h>

DS3231_Simple Clock;

#define BUTTONPIN 1
#define BUZZERPIN 4
#define RELAYPIN 3

void setup() {
  
pinMode(BUTTONPIN, INPUT);
pinMode(BUZZERPIN, OUTPUT);
pinMode(RELAYPIN, OUTPUT);

digitalWrite(BUTTONPIN, HIGH);
digitalWrite(BUZZERPIN, LOW);
digitalWrite(RELAYPIN, HIGH);

// Start the clock and clear any old timers
 Clock.begin();
 Clock.disableAlarms();

// Make variables for our static start time 
DateTime MyTimestamp;
  
  
  MyTimestamp.Day    = 12;
  MyTimestamp.Month  = 1;
  MyTimestamp.Year   = 19; 
  MyTimestamp.Hour   = 0;
  MyTimestamp.Minute = 0;
  MyTimestamp.Second = 0;
  
// Write that static start time to the clock  
  Clock.write(MyTimestamp);

 // How long should the timer last? (update the variable)
  MyTimestamp.Minute   = 1;                       
  
 // Set the type of alarm
  Clock.setAlarm(MyTimestamp, DS3231_Simple::ALARM_MATCH_MINUTE); 
  

}

void loop() {

  uint8_t AlarmsFired = Clock.checkAlarms();
  
  if(AlarmsFired & 2)
  {

    // Beep for 1 second
    digitalWrite(BUZZERPIN, HIGH);
    delay(1000);
    digitalWrite(BUZZERPIN,LOW);

    // Flash 3 times for 1 second each and turn off
    digitalWrite(RELAYPIN,LOW);
    delay(1000);
    digitalWrite(RELAYPIN,HIGH);
    delay(1000);
    digitalWrite(RELAYPIN,LOW);
    delay(1000);
    digitalWrite(RELAYPIN,HIGH);
    delay(1000);
    digitalWrite(RELAYPIN,LOW);
    delay(1000);
    digitalWrite(RELAYPIN,HIGH);
    delay(1000);
    digitalWrite(RELAYPIN,LOW);
  }
if(digitalRead(BUTTONPIN) == LOW){
  resettimer();
  return;
}
  delay(100);
}

void resettimer(){
  digitalWrite(RELAYPIN,HIGH);
Clock.disableAlarms();
DateTime MyTimestamp;
  
  
  MyTimestamp.Day    = 12;
  MyTimestamp.Month  = 1;
  MyTimestamp.Year   = 19; 
  MyTimestamp.Hour   = 0;
  MyTimestamp.Minute = 0;
  MyTimestamp.Second = 0;
  
  Clock.write(MyTimestamp);
 
  MyTimestamp.Minute   = 30;                       
  Clock.setAlarm(MyTimestamp, DS3231_Simple::ALARM_MATCH_MINUTE); 
}
```

fkkjohn, 2023-04-11:
A little bit closer (seemingly) with the digistump 1.6.7 board definitions and drivers.

```
Arduino: 1.8.18 (Windows 10), Board: "Digispark (16mhz - No USB)"

C:\Users\owner\Documents\Arduino\libraries\DS3231_Simple-master\DS3231_Simple.cpp: In member function 'void DS3231_Simple::promptForTimeAndDate(Stream&)':

C:\Users\owner\Documents\Arduino\libraries\DS3231_Simple-master\DS3231_Simple.cpp:755:12: error: 'class Stream' has no member named 'readBytes'

     Serial.readBytes(buffer, 2);

            ^

C:\Users\owner\Documents\Arduino\libraries\DS3231_Simple-master\DS3231_Simple.cpp:763:12: error: 'class Stream' has no member named 'readBytes'

     Serial.readBytes(buffer, 2);

            ^

C:\Users\owner\Documents\Arduino\libraries\DS3231_Simple-master\DS3231_Simple.cpp:771:12: error: 'class Stream' has no member named 'readBytes'

     Serial.readBytes(buffer, 2);

            ^

C:\Users\owner\Documents\Arduino\libraries\DS3231_Simple-master\DS3231_Simple.cpp:779:12: error: 'class Stream' has no member named 'readBytes'

     Serial.readBytes(buffer, 2);

            ^

C:\Users\owner\Documents\Arduino\libraries\DS3231_Simple-master\DS3231_Simple.cpp:787:12: error: 'class Stream' has no member named 'readBytes'

     Serial.readBytes(buffer, 2);

            ^

C:\Users\owner\Documents\Arduino\libraries\DS3231_Simple-master\DS3231_Simple.cpp:795:12: error: 'class Stream' has no member named 'readBytes'

     Serial.readBytes(buffer, 2);

            ^

C:\Users\owner\Documents\Arduino\libraries\DS3231_Simple-master\DS3231_Simple.cpp:804:12: error: 'class Stream' has no member named 'readBytes'

     Serial.readBytes(buffer, 1);

            ^

C:\Users\owner\Documents\Arduino\libraries\DS3231_Simple-master\DS3231_Simple.cpp:815:12: error: 'class Stream' has no member named 'readBytes'

     Serial.readBytes(buffer, 1);

            ^

exit status 1

Error compiling for board Digispark (16mhz - No USB).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

sleemanj, 2023-04-11:
Interesting, I can see that the problem is that the IDE (including my own 1.8.19) is pulling a different Wire library than the one provided by the core.

Will have to investigate.

fkkjohn, 2023-04-11:
Thanks. I'll let you know if I get it figured out. I'm also trying to recover my modules that are now missing bootloaders.

sleemanj, 2023-04-11:
Ok, it wasn't that actually.

in DS321_Simple.h

Below where it says `#include <Wire.h>`

Add a line `#include <Stream.h>`

This should compile your script correctly (provided you selected an appropriate core for Tiny85).

Thge issue is that the Wire library for the Tiny85 does not, or no longer does, inherit from Stream (why I don't know, didn't write it), where other chip's Wire library does.

fkkjohn, 2023-04-11:
I confirm that it is now compiling and uploading, and the button is still being registered, but the alarm either never fires or it never reports as having fired when polled. We're back to where I was yesterday afternoon, minus the later trouble with the bootloaders being wiped. Still, it's progress! I'll experiment a bit more, and let you know what I find. I'm currently using my previously working hardware to verify whether compiled programs actually work. What I can tell you is that the relay clicks a bit while turning on (good I think), then it doesn't seem to do anything once everything settles down to the correct pin settings. The button goes high, the relay is off, and the buzzer isn't buzzing. All as it should be, but I'm not getting any action after the 1 minute mark.

sleemanj, 2023-04-11:
I would suggest you try a simplified test case, for instance the alarms example.

fkkjohn, 2023-04-11:
You're right. I don't have serial out, but I can tell the onboard led to light up and turn off for varying amounts of time, so it's definitely working. I'll figure out what's wrong with my relays with just a little effort I'm sure. Thank you very much for your help. Problems solved.

Time to clean up all of my investigative efforts. =) Thanks again.

closed 2023-04-11 by fkkjohn

fkkjohn, 2023-04-12:
Just thought I'd stop by and let you know that I figured out the problem that kept my alarm from triggering. The reset switch was wired through PB1 as an input, and even after attaching a pull up resistor, the built in LED would drop the voltage enough that it never went high. It was in a constant loop of resetting itself. I removed the LED and problem solved. Thanks for all your help getting everything compiling. You're doing a great service by sharing your skills and keeping your libraries updated. Thank you.