sleemanj/DS3231_Simple #26: Error - Import library on Arduino IDE 2.2.1

RogerLTx, 2023-09-22:
Steps followed: Sketch\Include Library\Add .ZIP Library and the error message appears:
"Error 13 Internal. Library installed failed: archive is not valid: multiple files found in ZIP file top level."

sleemanj, 2023-09-23:
Try the attached
[DS3231_Simple.zip](https://github.com/sleemanj/DS3231_Simple/files/12705321/DS3231_Simple.zip)

closed 2023-09-23 by sleemanj

RogerLTx, 2023-09-24:
> Enviada em: sexta-feira, 22 de setembro de 2023 22:25

Hi.
Thank you for your answer.
Same error
***@***.***

***@***.***

Rogerio

sleemanj, 2023-09-24:
Sorry error in previous zip archive, this one is tested in 2.2.1 working
[DS3231_Simple.zip](https://github.com/sleemanj/DS3231_Simple/files/12708191/DS3231_Simple.zip)

RogerLTx, 2023-09-25:
Oh! Don’t be sorry.
I appreciate your patience and support.
Everything is working now!

Thank you.

Rogerio

Enviada em: domingo, 24 de setembro de 2023 06:41