sleemanj/DS3231_Simple #27: Ds3231 HelloWorld example not properly working

Fusionlight13, 2023-10-08:
I just imported this library for the first time and started the basic HelloWorld script. When I run it and answer the first two prompts, I get two more prompts consecutively. I try to answer them all separately but the loop just goes back to the first day of month input.

sleemanj, 2023-10-08:
Ensure you are entering in exactly the format specified by the prompt (ie, for day enter 09 not 9).  Set line ending to Newline

Code for the prompts is here:
https://www.sbs.com.au/nitv/article/why-blak-not-black-artist-destiny-deacon-and-the-origins-of-this-word/7gv3mykzv

closed 2023-10-08 by sleemanj

Fusionlight13, 2023-10-09:
Unfortunately I could not get it to work. I padded the numbers and made sure I put it in correctly several times but still It skips input. The baud rate is right and its set to newline. Tried installing it a second time but the results are the same.

sleemanj, 2023-10-09:
I would suggest you may actually have a hardware issue which is causing a reset.

After the Serial.begin line, add 

    Serial.println("Starting Up...");

Fusionlight13, 2023-10-09:
I ran the SetDateTime sketch and it seems to work fine.  It's remembering it when I restart it with the clock. write() line deleted. It's very strange. I guess I just have to set it manually without the Clock.promptForTimeAndDate() method.